mbed-ce/mbed-os #441: Failed to program flash using OpenOCD on a STM32F4 target board

wdx04, 2025-03-24:
Hi,

I just tried programming my STM32F401 board with a cheap CMSIS-DAP debugger(via the flash-mbed-app UTILITY target in the VS Code IDE, Windows 11 x64).
But I'm getting the following error:

```
[build] [1/1 100% :: 1.114] Flashing mbed-app with OpenOCD...
[build] xPack Open On-Chip Debugger 0.12.0+dev-01850-geb6f2745b-dirty (2025-02-07-10:08)
[build] Licensed under GNU GPL v2
[build] For bug reports, read
[build] 	http://openocd.org/doc/doxygen/bugs.html
[build] srst_only separate srst_nogate srst_open_drain connect_deassert_srst
[build] DEPRECATED! use 'gdb port', not 'gdb_port'
[build] Info : CMSIS-DAP: SWD supported
[build] Info : CMSIS-DAP: FW Version = 2.0.0
[build] Info : CMSIS-DAP: Interface Initialised (SWD)
[build] Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 0 TDO = 0 nTRST = 0 nRESET = 1
[build] Info : CMSIS-DAP: Interface ready
[build] Info : clock speed 2000 kHz
[build] Info : SWD DPIDR 0x2ba01477
[build] Info : [stm32f4x.cpu] Cortex-M4 r0p1 processor detected
[build] Info : [stm32f4x.cpu] target has 6 breakpoints, 4 watchpoints
[build] Info : [stm32f4x.cpu] Examination succeed
[build] Info : [stm32f4x.cpu] gdb port disabled
[build] [stm32f4x.cpu] halted due to undefined, current mode: Thread 
[build] xPSR: 0x21000000 pc: 0x080024c4 psp: 0x20001748
[build] ** Programming Started **
[build] Info : device id = 0x00016423
[build] Info : flash size = 256 KiB
[build] Info : flash size = 512 bytes
[build] Warn : no flash bank found for address 0x00000000
[build] ** Programming Finished **
[build] ** Resetting Target **
[build] shutdown command invoked
[driver] Build completed: 00:00:01.155
[build] Build finished with exit code 0
```

Running the following command in a terminal gives the same error:

```
openocd -f stm32f401cc.cfg -c "program mbed-app.bin reset exit"
```

And there is no problem with the following two commands:

```
openocd -f stm32f401cc.cfg -c "program mbed-app.hex reset exit"
openocd -f stm32f401cc.cfg -c "program mbed-app.elf reset exit"
```

Also, appending 0x08000000 in UploadMethodOPENOCD.cmake, line 53 also works(see https://openocd.org/doc/html/Flash-Programming.html): 

```
		-c "program ${BINARY_FILE} reset exit 0x08000000"
```

So, the build system sends mbed-app.bin (no address information) to OpenOCD, and OpenOCD tries to write the file into an incorrect address "0x00000000".

Is it possible to program the .hex or .elf file, instead of the .bin file with the build system?
Or can we specify the target address in the OpenOCD configuration file (.cfg in openocd_cfgs folder)?

JohnK1987, 2025-03-24:
Hello, 

I remember a PR few months ago - https://github.com/mbed-ce/mbed-os/pull/331
So if I understand this
https://github.com/mbed-ce/mbed-os/blob/1fdce7d1157c136eb8998cec8fc8c35ddfbd4020/tools/cmake/UploadMethodManager.cmake#L142-L155

try set MBED_OUTPUT_EXT to 'hex' in your top level CMakeLists.txt

wdx04, 2025-03-25:
Hi John,

Setting MBED_OUTPUT_EXT to 'hex' works fine for me.
Thanks.

closed 2025-03-25 by wdx04