mbed-ce/mbed-os #460: Documentation fork

MariwanJ, 2025-05-07:
Dear developers, 

I thank you that you put time on this project. I love MbedOS and wish that the project survive.
I would like to ask you to make a fork of the documentation so we don't loose the information. Even if there are tons of error (Guessing) but still it is good to preserve it. 

Hope you will have time to do so. 

Thanks,
Mariwan

multiplemonomials, 2025-05-13:
You have a very good point! I have been thinking about the same thing. Just this week we set up a new, rudimentary doc website: https://mbed-ce.dev . It's driven by this repo: https://github.com/mbed-ce/mbed-ce-dot-dev. I am hoping that over the next months we can start migrating documentation pages over!

MariwanJ, 2025-05-13:
@multiplemonomials 
Great. Thanks. I hope I would be able to contribute to this wonderful project.  You might know that I forked the hello world project and made a pdf that explain how to use Eclipse for developing Mbed-OS CE project.